WOA-Project/WOA-Deployer-Lumia #6: Error 995 when injecting drivers

jhoff80, 2019-02-14:
I tried to deploy an image twice using WOA Deployer 2.0.0, and both times got an error 995 during the second round of injecting drivers.  Not sure if any steps are supposed to take place after injecting drivers finishes, but it just halts the process.  The phone is bootable after this though.

Checking the log shows that an error takes place with the following driver:

2019-02-13 01:14:04.501 -05:00 [Verbose] Installing 40 of 71 - D:\WOA Deployer for Lumia (Good UEFI Patched)\Gui\Downloaded\MSM8994-8992-NT-ARM64-Drivers-master\msm8994\Pre-OOBE\System\QCListenSoundModel8994.inf: Error - An error occurred. The driver package could not be installed. 
2019-02-13 01:14:04.501 -05:00 [Verbose] For more information, check for log files in the <windir>\inf folder of the target image.

This is with an 1809 install.wim.

CornyjK, 2019-02-14:
Try again and comment results.

closed 2019-02-14 by CornyjK

xerone9, 2019-03-25:
Same. Installed windows 10 arm on lumia 950 with deployer 2.4. Received same error (regarding driver installation failed) and windows is still bootable.