WOA-Project/WOA-Deployer-Lumia #11: Driver update with the WOA deployer

PatjedG, 2019-02-17:
Is it possible to make a function in the WOA deployer to update the drivers automatic with the latest drivers on github.
Would be nice so that you don't have to do a reinstallation of the whole oparating system.
I saw that in version 2.0 the tool automatic downloads the latest drivers and injecting it in the prefase, maby it is possible to expand that as a sepparate function to update trough mass storage if there are updates availeble

CornyjK, 2019-02-18:
Read FAQ.

SuperJMN, 2019-02-18:
@PatjedG I'll make that possible. Check the Scripts folder in you WOA Deployer. You can already tweak the script to do what you want. I'm sure you will get it soon :)

SuperJMN, 2019-02-18:
About the drivers, I'm not sure it it's possible with most of them, since there are a lot of essential drivers that cannot be replaced once Windows has been installed.

closed 2019-02-18 by SuperJMN