WOA-Project/WOA-Deployer-Lumia #12: Deploying fails at enar completion with DISM error 2

Redn2000, 2019-03-01:
**DISM ERROR 2**
After attemtping several times on my Lumia 950 XL with the most recent Windows 10 ARM build "18277" I am unable to finish flashing the build after driver injection and receive a "DISM exited with code 2" I'm unsure if there was a driver I had missed during the preparation for this, but given that I've otherwise used WPInternals several times without issue, it may be that a driver is missing or fails to download somewhere along the line.

**Ending log excerpt**
```
019-02-28 18:55:07.880 -05:00 [Verbose] Error: 2
2019-02-28 18:55:07.880 -05:00 [Verbose] 
2019-02-28 18:55:07.880 -05:00 [Verbose] Cannot find the specified file 'Downloaded\MSM8994-8992-NT-ARM64-Drivers-master\msm8994\Pre-OOBE'
2019-02-28 18:55:07.880 -05:00 [Verbose] 
2019-02-28 18:55:07.880 -05:00 [Verbose] The DISM log file can be found at C:\WINDOWS\Logs\DISM\dism.log
2019-02-28 18:55:07.886 -05:00 [Error] An error has occurred
Deployer.Exceptions.DeploymentException: There has been a problem during deployment: DISM exited with code 2.
   at Deployer.NetFx.ImageServiceBase.<InjectDrivers>d__3.MoveNext() in D:\a\1\s\Source\DeployerPlatform\Deployer.NetFx\ImageServiceBase.cs:line 74
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Deployer.Tasks.InjectDrivers.<Execute>d__4.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Deployer.Execution.ScriptRunner.<Run>d__5.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Deployer.Execution.ScriptRunner.<Run>d__4.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Deployer.Lumia.WoaDeployer.<Deploy>d__5.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Deployer.Lumia.Gui.ViewModels.DeploymentViewModel.<Deploy>d__9.MoveNext() in D:\a\1\s\Source\Deployer.Lumia.Gui\ViewModels\DeploymentViewModel.cs:line 55
2019-02-28 18:55:07.887 -05:00 [Information] Error: There has been a problem during deployment: DISM exited with code 2.

```

CornyjK, 2019-03-05:
Delete your `Downloaded` folder and try again.

Redn2000, 2019-03-08:
It looks like the real issue was caused by it not being run from my main C: drive, which for some reason didn't give it the right permissions even when being run as an administrator.

closed 2019-03-08 by Redn2000