WOA-Project/WOA-Deployer-Lumia #20: Error: Phone disk preparation failed. Cannot prepare the phone for the deployment

pierrejoubert, 2019-04-10:
When attempting to perform a deployment, I get the following error:

**Error: Phone disk preparation failed. Cannot prepare the phone for the deployment**

I have attached the log, I first got this with a brand new deployment after flashing the phone with the original rom.  It refuses to progress beyond setting up the Partitions.

Phone:  Lumia 950XL

[Log-20190410_001.txt](https://github.com/WOA-Project/WOA-Deployer-Lumia/files/3061491/Log-20190410_001.txt)

SuperJMN, 2019-04-10:
Thanks for reporting! Please, try with the new release https://github.com/WOA-Project/WOA-Deployer-Lumia/releases/tag/v2.4.2

pierrejoubert, 2019-04-12:
Awesome, thanks, works well, no more issues. :)

closed 2019-04-12 by pierrejoubert

FrKaram, 2019-04-12:
In my case, 2.4.1 does work, 2.4.2 generates the message "Phone disk preparation failed. Cannot prepare the phone for the deployment"

dynamicdev, 2019-04-13:
I am trying to dual boot my Lumia 950 XL. All went well until I faced "Phone disk preparation failed. Cannot prepare the phone for the deployment" error.

Now my phone is not booting up... Microsoft logo appears and nothing happens after that... Tried to put in Flash mode... Hard reset... all in vein...

Any guess... What went wrong...