WOA-Project/WOA-Deployer-Lumia #23: Phone bricked: "Phone disk preparation failed. Cannot prepare the phone for the deployment"

dynamicdev, 2019-04-13:
I am trying to dual boot my Lumia 950 XL. All went well until I faced "Phone disk preparation failed. Cannot prepare the phone for the deployment" error.

Now my phone is not booting up... Microsoft logo appears and nothing happens after that... Tried to put in Flash mode... Hard reset... all in vein...

Any guess... What went wrong...

CornyjK, 2019-04-13:
Do not double-post. Send us log, without log we can not provide any further help. Are you running latest release version?

dynamicdev, 2019-04-13:
[Log-20190413.txt](https://github.com/WOA-Project/WOA-Deployer-Lumia/files/3076265/Log-20190413.txt)
Logs attached... 

I am using Lumia V2.4.2
[image]

dynamicdev, 2019-04-13:
[image]

Error on phone

SuperJMN, 2019-04-13:
I'll try to reproduce this issue later. In order to fix this, put the phone into Mass Storage Mode with either Developer Menu or with WPInternals and redeploy. If it fails preparing the partition layout, retry until it works.

dynamicdev, 2019-04-13:
My phone is not coming in Mass Storage Mode. WPInternals tool is not connecting as due to "Waiting for connection with phone..."...

Please guide me how to phone in Developer Menu.

SuperJMN, 2019-04-13:
Inside WPInternals, click on Manual section and click on "interrupt the boot process", then reboot the phone. After that, you will be able to switch to Mass Storage Mode.

Obtener Outlook para Android<https://aka.ms/ghei36>

dynamicdev, 2019-04-13:
Thanks Got MSM...

Seems there is problem with Storage as per logs but in my compute DATA partition there is 20 GB space available while 1 GB space on MainOS partition...

[image]

SuperJMN, 2019-04-13:
OK, it looks like it didn't resize your Data partition. Run WOA Deployer again and see if it works.

Obtener Outlook para Android<https://aka.ms/ghei36>

[WhatsApp Image 2019-04-13 at 7 43 31 PM]<https://user-images.githubusercontent.com/4348192/56080899-7964e600-5e24-11e9-97a9-989aa9be8fc7.jpeg>

dynamicdev, 2019-04-13:
This time it worked. But I think problem is somewhere else. 

When I tried Windows 10 ARM 1903 version, it failed...
[image]

but when tried 1809 version it worked...
[image]

Is this could be a problem?

dynamicdev, 2019-04-13:
Windows image deployment done but phone is not booting to OS selection menu.

I can see "Hello from San Francisco" screen...
[Log-20190413.txt](https://github.com/WOA-Project/WOA-Deployer-Lumia/files/3076481/Log-20190413.txt)

What next should I do?

SuperJMN, 2019-04-13:
Reboot until it passes that screen. That's not a problem of WOA Deployer, but the UEFI, that gets stuck sometimes.

dynamicdev, 2019-04-13:
Loading drive at .... EntryPoint ... bootmgfu.efi STUCK here

Rebooted more than 10 times... Not moving further

SuperJMN, 2019-04-13:
Try build 18865 to see if you get the same issue.

Obtener Outlook para Android<https://aka.ms/ghei36>

dynamicdev, 2019-04-13:
[image]

Reboot loop...

dynamicdev, 2019-04-14:
I tried 18362 (Release build) before your suggestion... Phone rebooted but became blank while "**Getting ready**"... It seems battery drained...

Now phone is not able to setup. Its asking to "Install a driver anyway" and tons of different behavior.

Also, I can not see boot menu, when phone reboots it directly goes to Windows Installation process... Does not ask me to select an OS option... Getting frustrated...

Now trying 18865...

dynamicdev, 2019-04-14:
Installed 18865...

Now I reached "Lets start with a region. Is this right?". From here I am not able to select anything... Seems touch is not working...

SuperJMN, 2019-04-14:
Touch usually fails to works. It's another known issue. Please, reboot until it starts working.

dynamicdev, 2019-04-14:
Finally, I got boot menu and my Windows Phone OS back. Still Windows 10 OS not working.

I AM DONE.

closed 2019-04-14 by dynamicdev

qufanli, 2019-06-15:
Hello!
Want to ask 950xl brush Windows sarm, install GPU driver after frequent crash reboot, any version has tried! I don't know where the problem is.