WOA-Project/WOA-Deployer-Lumia #28: Phone disk preparation failed. Cannot prepare the phone for the deployment

tkiapril, 2019-05-07:
**Describe the bug**
When deploying 17134 to Lumia 950XL with Deployer 2.5.0 on Windows 10, "Phone disk preparation failed. Cannot prepare the phone for the deployment" appears on Formatting partitions (SYSTEM)

**To Reproduce**
Deploy as usual on 2.5.0.

**Expected behavior**
Deploy normally.

**Log file**

[Log-20190507.txt](https://github.com/WOA-Project/WOA-Deployer-Lumia/files/3151817/Log-20190507.txt)

I am well aware that similar problems have been posted as an issue several times, but I cannot get past this no matter how many times I try. (including reboots)

tkiapril, 2019-05-07:
[image]

Are these files supposed to be in EFIESP partition? I have an impression that directories such as Program Files are not supposed to be in EFIESP, since it's a partition for storing efi specific binaries and configurations.

dreimer1986, 2019-05-08:
I have the same folders and it worked fine up to now, so... I guess the answer is yes

tkiapril, 2019-05-09:
I tried to restore my device to factory and re-unlock the bootloader; the files and directories are still there, so this would be irrelevant in this regard.

I just realized @SuperJMN bumped the platform up some commit which includes alternate formatting method. The release is not published yet, but I could access the CI at https://superjmn.visualstudio.com/WOA-Deployer/_build/results?buildId=411 and using it seems to work. If someone is having the same problem, I would suggest trying the artifact from the link to see if it works for them. Closing issue.

SuperJMN, 2019-05-09:
Hey @tkiapril, could you please try the build here? https://dev.azure.com/superjmn/WOA-Deployer/_build/results?buildId=413

I have refined the code and I would like to know if it's still working, since I cannot reproduce the issue myself. Thanks in advance :)

tkiapril, 2019-05-09:
@SuperJMN Will do!

tkiapril, 2019-05-09:
@SuperJMN It seems like that it won't work. Log here:  [Log-20190509.txt](https://github.com/WOA-Project/WOA-Deployer-Lumia/files/3161118/Log-20190509.txt)

Specifically I don't think it's entering the https://github.com/WOA-Project/WOA-Deployer-Platform/blob/master/Deployer.NetFx/DiskApi.cs#L56-L60 block; I don't get "Cannot get volume. Assigning a driver letter so Windows creates a volume for the partition" message in my log.

8d6cb8c7cde9e886d00cb3a3f6b5218660fd5d22 worked like a charm (in the log I can see "2019-05-09 01:46:28.495 -07:00 [Debug] Format failed. Attempting alternate method.")

SuperJMN, 2019-05-09:
Fixed with https://github.com/WOA-Project/WOA-Deployer-Lumia/commit/055c5e1de25361db7cf6a2bd0d88d5b89d8fa379

Version 2.5.1 :)

closed 2019-05-09 by SuperJMN

Yelkambe, 2025-01-11:
> @SuperJMN It seems like that it won't work. Log here: [Log-20190509.txt](https://github.com/WOA-Project/WOA-Deployer-Lumia/files/3161118/Log-20190509.txt)
> 
> Specifically I don't think it's entering the https://github.com/WOA-Project/WOA-Deployer-Platform/blob/master/Deployer.NetFx/DiskApi.cs#L56-L60 block; I don't get "Cannot get volume. Assigning a driver letter so Windows creates a volume for the partition" message in my log.
> 
> [8d6cb8c](https://github.com/WOA-Project/WOA-Deployer-Lumia/commit/8d6cb8c7cde9e886d00cb3a3f6b5218660fd5d22) worked like a charm (in the log I can see "2019-05-09 01:46:28.495 -07:00 [Debug] Format failed. Attempting alternate method.")

Yelkambe, 2025-01-11:
please kindly help me i have a similar problem of phone disk patition