WOA-Project/WOA-Deployer-Lumia #29: [950XL] 18362 Unexpected restart loop on first time setup causing setup failure.

raidensnake, 2019-05-08:
I tried installing 18362.86 however I get an error trying to start it, "The computer started unexpectedly or encountered an unexpected error. Windows installation cannot proceed. To install windows, click "OK" to restart the computer, then restart the installation." It just shows that message every time I try to restart it.

BenNottelling, 2019-05-09:
You shutdown your phone before the install finished, reinstall Windows and try again. After the screen goes black wait and eventually you should see windows loading and starting to install, if you turn it off before it's done you'll get that error message and have to start again

raidensnake, 2019-05-09:
Sorry but I didn't touch it after I rebooted into normal mode. it to a blank screen during the boot then it rebooted itself after just 2 mins.

BenNottelling, 2019-05-09:
Try again then, it's not a perfect process. If that doesn't work try a different build, I'll tell you the build I used later if that doesn't work for you

raidensnake, 2019-05-09:
I also tried 17763.475 and it did the exactly the same thing.

BenNottelling, 2019-05-09:
[image]
Did this yesterday, took a few attempts, but it does work.
How is the battery on your device? A bad battery can cause random reboots and shut downs.

tkiapril, 2019-05-10:
Logs? The developers can't help you if they don't know what's going south in your deployment.

raidensnake, 2019-05-10:
Well that's the thing. The deployment was successful yet the install fails on the phone itself.

tkiapril, 2019-05-10:
Your best bet still is to upload the logs. The logs contain the nitty gritty details about the deployment, and it's not just pass/fail like if it's either way or other. I'm not a dev for this project, but without the log files the devs will never know what's going wrong in your case and won't be able to help you.

BaRRaKudaRain, 2019-05-10:
Maybe you should try another Windows 10 build? 16299 RS3 for example. Later
builds may cause some problems.

raidensnake, 2019-05-10:
[Log-20190509.txt](https://github.com/WOA-Project/WOA-Deployer-Lumia/files/3166756/Log-20190509.txt) Here's the last log.

carues, 2019-05-10:
I had the same message with builds 
creatingISO_18362.86_pt-br_arm64_professional.cmd and creatingISO_17763.349_pt-br_arm64_Professional.cmd . The error shows in the phone screen after applied the windows image as shown in the attached file (Is in Brazilian Portuguese) .

[image]

carues, 2019-05-10:
Where I find the logs in the cellphone to post here?

raidensnake, 2019-05-11:
+1 on @carues The deploy logs themselves don't show any issues.

raidensnake, 2019-05-11:
[deploy.txt](https://github.com/WOA-Project/WOA-Deployer-Lumia/files/3168655/deploy.txt)
There's the recent deploy log.

SuperJMN, 2019-05-11:
The deployment was successful. That's why the log doesn't have any error inside.

SuperJMN, 2019-05-11:
BTW, I'm closing this issue since it's not Deployer related, but it faills into the  WOA or Drivers or EUFI side. Sorry guys. Deployer responsibility is to deploy, only.

SuperJMN, 2019-05-11:
One final comment. The boot loop is normally a driver issue that can be fixed redeploying. Try it to see if that helps :)

closed 2019-05-11 by SuperJMN

raidensnake, 2019-05-11:
I tried that several times @SuperJMN and it made no difference whatsoever.

SuperJMN, 2019-05-11:
Can you tried with another build, for example 18890 or 18894?

raidensnake, 2019-05-11:
I tried those as well and got the same problem.

carues, 2019-05-11:
I tried one more time with build creatingISO_17763.349_pt-br_arm64_Professional.cmd but in this time I didn't disconnect the cellphone from the computer as the Woa.deployed says after end the process. And it works!!

Obter o Outlook para Android<https://aka.ms/ghei36>

BaRRaKudaRain, 2019-05-11:
Seems that WOA deployer deployed the installer image, not the install.wim file. So that's the reason that it cannot continue.

An outsider, the BaRRaKudaRain

SuperJMN, 2019-05-11:
With normal usage, that cannot happen. The wim selecting dialog prompts specifically for “install.wim”, that IS the correct image.

tkiapril, 2019-05-11:
@BaRRaKudaRain What I see in your logs is that you are using uup converter to download the files and convert them. Try using uup-dump available on gitlab if you aren't already. (I cannot link them here because I'm not sure about the legality, sorry.)

tkiapril, 2019-05-11:
Btw, I guess that it is extremely unlikely that it might be the case, but make sure you use sources/install.wim, not boot.wim.

tkiapril, 2019-05-11:
@BaRRaKudaRain sorry, seems like I mentioned the wrong person. That comment was supposed to go to @raidensnake .