WOA-Project/WOA-Deployer-Lumia #30: Phone disk preparation failed." while formatting SYSTEM for v2.5.0

chenkaigithub, 2019-05-09:
Install windows 10 on arm for 950 failed.
Use windows 10 version:
18362.86.190423-1423.19H1_RELEASE_SVC_PROD3_CLIENTPRO_OEMRET_A64FRE_ZH-CN.ISO

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Log file**
Please put your log file here (just drag&drop it).
[Log-20190509.txt](https://github.com/WOA-Project/WOA-Deployer-Lumia/files/3161127/Log-20190509.txt)

tkiapril, 2019-05-09:
Possible duplicate issue #28, try CI build 411: https://dev.azure.com/superjmn/WOA-Deployer/_build/results?buildId=411 (Download from Artifacts->WOA Deployer)

SuperJMN, 2019-05-09:
As @tkiapril says, please try the build he's linking. If you don't know how to do it, please, comment :)

SuperJMN, 2019-05-09:
Reverted the changes. This should work: 2.5.1 https://github.com/WOA-Project/WOA-Deployer-Lumia/commit/055c5e1de25361db7cf6a2bd0d88d5b89d8fa379

closed 2019-05-09 by SuperJMN

BaRRaKudaRain, 2019-05-09:
Maybe you should try older build?

chenkaigithub, 2019-05-10:
I bought an old phone to install, but it is now brick. I found about 40 partitions on the disk. So I think this issue could be raised by this partitions which is shown on 0 M.  I removed some and keep the system partitions. But the phone cannot be started anymore and the screen is always black. I am not it's a software issue or hardware issue now.

BenNottelling, 2019-05-10:
It's not bricked just use the Microsoft device recovery tool

chenkaigithub, 2019-05-10:
MDRT cannot detect device.

chenkaigithub, 2019-05-10:
[image]

chenkaigithub, 2019-05-10:
Finally, the phone is back. But I have the same problem.

chenkaigithub, 2019-05-10:
2.5.1 is used

chenkaigithub, 2019-05-10:
New log is attached.
[Log-20190510.txt](https://github.com/WOA-Project/WOA-Deployer-Lumia/files/3165024/Log-20190510.txt)

tkiapril, 2019-05-10:
Mine works with 2.5.1, so apparently it's a different issue than mine. I'm wondering if your host fails to assign a drive letter for the device, but I cannot be sure since the log file doesn't say anything about it.

chenkaigithub, 2019-05-13:
I used another windows machine to install. It's workable. I attached the log. But now wifi/audio/Bluetooth
 is still not workable.
[Log-20190513.txt](https://github.com/WOA-Project/WOA-Deployer-Lumia/files/3171213/Log-20190513.txt)

tkiapril, 2019-05-13:
Are you using a 950 non-XL (Talkman)? As per https://github.com/WOA-Project/MSM8994-8992-NT-ARM64-Drivers/blob/1905/README.md audio and Wi-Fi is nonfunctional for 950(MSM8992) at the moment, and will not be supported very soon as well since devs are more engaged with MSM8994.

CornyjK, 2019-05-13:
You are using Talkman (950), which does not have those features working

chenkaigithub, 2019-05-14:
I think format partition issue is from PowerCli. It occurs error as bellow.
[image]

[image]

[image]

I used another win 10 , it's workable.
[image]

[image]

chenkaigithub, 2019-05-14:
> You are using Talkman (950), which does not have those features working

Thank you. Hope Talkman could be full supported ASAP.