WOA-Project/WOA-Deployer-Lumia #31: Could not locate file

fangguangyang, 2019-05-10:
2019-05-07 21:37:23.050 +08:00 [Verbose] Started WOA Deployer for Lumia v2.5.0
2019-05-07 21:37:30.506 +08:00 [Verbose] Checking path: "" does not exist
2019-05-07 21:38:29.091 +08:00 [Verbose] Checking path: "" does not exist
2019-05-07 21:38:34.012 +08:00 [Verbose] Default directory for WimFolder has been set to "H:\sources"
2019-05-07 21:38:34.012 +08:00 [Verbose] Trying to load WIM metadata file at '"H:\sources\install.wim"'
2019-05-07 21:38:34.059 +08:00 [Verbose] Getting WIM stream
2019-05-07 21:38:34.059 +08:00 [Verbose] (WIM) Finding Magic Bytes...
2019-05-07 21:38:34.059 +08:00 [Verbose] (WIM) Found Magic Bytes at 0
2019-05-07 21:38:34.059 +08:00 [Verbose] (WIM) Finding WIM XML Data...
2019-05-07 21:38:34.059 +08:00 [Verbose] (WIM) Found WIM XML Data at 046011288642
2019-05-07 21:38:34.075 +08:00 [Verbose] Wim metadata deserialized correctly WimMetadata { TotalBytes: "4601128864", Images: [ImageMetadata { DirectoryCount: "24347", FileCount: "103628", TotalBytes: "17695516416", HardLinkBytes: "6834111914", CreationTime: Time { HighPart: "0x01D44D1A", LowPart: "0x1C0774BF" }, LastModificationTime: Time { HighPart: "0x01D5035D", LowPart: "0xE718C29B" }, WimBoot: "0", Windows: Windows { Arch: "12", ProductName: "Microsoft® Windows® Operating System", EditionId: "Professional", InstallationType: "Client", ServicingData: ServicingData { GdrDuRevision: "0", PKeyConfigVersion: "10.0.17763.1;2016-01-01T00:00:00Z" }, ProductType: "WinNT", ProductSuite: "Terminal Server", Languages: Languages { Language: "zh-CN", Default: "zh-CN" }, Version: Version { Major: "10", Minor: "0", Build: "17763", SpBuild: "349", SpLevel: "0" }, SystemRoot: "WINDOWS" }, Name: "Windows 10 Pro", Description: "Windows 10 Pro", Flags: null, DiplayName: "Windows 10 Pro", DisplayDescription: "Windows 10 Pro", Index: "1" }] }
2019-05-07 21:38:34.091 +08:00 [Verbose] WIM metadata file at '"H:\sources\install.wim"' retrieved correctly
2019-05-07 21:39:31.443 +08:00 [Information] # Starting deployment...
2019-05-07 21:39:31.459 +08:00 [Information] Deleting previously downloaded deployment files
2019-05-07 21:39:35.218 +08:00 [Debug] Initializing Device Stream...
2019-05-07 21:39:35.231 +08:00 [Debug] Reading device GPT...
2019-05-07 21:39:35.234 +08:00 [Debug] Found DPP
2019-05-07 21:39:35.234 +08:00 [Debug] Found SBL1
2019-05-07 21:39:35.234 +08:00 [Debug] Found PLAT
2019-05-07 21:39:35.234 +08:00 [Debug] Reading SBL1 Partition
2019-05-07 21:39:35.235 +08:00 [Debug] Starting reading operation
2019-05-07 21:39:35.235 +08:00 [Debug] Reading from 54525952 to 55574016
2019-05-07 21:39:35.854 +08:00 [Debug] Reading pconf.bin from file system...
2019-05-07 21:39:35.897 +08:00 [Debug] Reading product.dat from file system...
2019-05-07 21:39:35.902 +08:00 [Debug] Is the device a MMO device: True
2019-05-07 21:39:35.907 +08:00 [Debug] Finished in: 0.688559 seconds.
2019-05-07 21:39:35.907 +08:00 [Debug] Drumroll...
2019-05-07 21:39:35.907 +08:00 [Debug] //////////////////
2019-05-07 21:39:35.907 +08:00 [Debug] product.dat
2019-05-07 21:39:35.907 +08:00 [Debug] //////////////////
2019-05-07 21:39:35.909 +08:00 [Debug] 42-7D-8F-D5-A7-F2-27-82-0D-5B-11-BF-8C-6F-76-70-C0-A0-62-2C-C6-1B-A9-5A-AE-E1-8F-75-17-FC-0B-77
2019-05-07 21:39:35.918 +08:00 [Verbose] "Model: Cityman, Variant: DualSim" detected
2019-05-07 21:39:35.978 +08:00 [Information] 
2019-05-07 21:39:35.978 +08:00 [Information] # Downloading resources
2019-05-07 21:39:36.109 +08:00 [Information] Fetching from Azure DevOps: LumiaWOA;Boot Shim;3;UEFI Loader for 8992 and 8994
2019-05-07 21:39:41.718 +08:00 [Information] Fetching asset MSM8994.UEFI.Lumia.950.XL.zip from latest release at https://github.com/WOA-Project/Lumia950XLPkg
2019-05-07 21:43:41.654 +08:00 [Error] An error has occurred
System.IO.IOException: 无法从传输连接中读取数据: 你的主机中的软件中止了一个已建立的连接。。 ---> System.Net.Sockets.SocketException: 你的主机中的软件中止了一个已建立的连接。
   在 System.Net.Sockets.Socket.EndReceive(IAsyncResult asyncResult)
   在 System.Net.Sockets.NetworkStream.EndRead(IAsyncResult asyncResult)
   --- 内部异常堆栈跟踪的结尾 ---
   在 System.Net.ConnectStream.EndRead(IAsyncResult asyncResult)
   在 System.IO.Stream.<>c.<BeginEndReadAsync>b__43_1(Stream stream, IAsyncResult asyncResult)
   在 System.Threading.Tasks.TaskFactory`1.FromAsyncTrimPromise`1.Complete(TInstance thisRef, Func`3 endMethod, IAsyncResult asyncResult, Boolean requiresSynchronization)
--- 引发异常的上一位置中堆栈跟踪的末尾 ---
   在 System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   在 System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   在 System.Net.Http.HttpClientHandler.WebExceptionWrapperStream.<ReadAsync>d__4.MoveNext()
--- 引发异常的上一位置中堆栈跟踪的末尾 ---
   在 System.Reactive.PlatformServices.ExceptionServicesImpl.Rethrow(Exception exception)
   在 System.Reactive.ExceptionHelpers.ThrowIfNotNull(Exception exception)
   在 System.Reactive.Subjects.AsyncSubject`1.GetResult()
   在 Deployer.Downloader.<Download>d__3.MoveNext()
--- 引发异常的上一位置中堆栈跟踪的末尾 ---
   在 System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   在 System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   在 Deployer.Downloader.<GetStream>d__5.MoveNext()
--- 引发异常的上一位置中堆栈跟踪的末尾 ---
   在 System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   在 System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   在 Deployer.Tasks.FetchGitHubLatestReleaseAsset.<Execute>d__9.MoveNext()
--- 引发异常的上一位置中堆栈跟踪的末尾 ---
   在 System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   在 System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   在 Deployer.Execution.ScriptRunner.<Run>d__5.MoveNext()
--- 引发异常的上一位置中堆栈跟踪的末尾 ---
   在 System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   在 System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   在 Deployer.Execution.ScriptRunner.<Run>d__4.MoveNext()
--- 引发异常的上一位置中堆栈跟踪的末尾 ---
   在 System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   在 System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   在 Deployer.Lumia.WoaDeployer.<Deploy>d__7.MoveNext()
--- 引发异常的上一位置中堆栈跟踪的末尾 ---
   在 System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   在 System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   在 Deployer.Lumia.Gui.ViewModels.DeploymentViewModel.<Deploy>d__11.MoveNext() 位置 D:\a\1\s\Source\Deployer.Lumia.Gui\ViewModels\DeploymentViewModel.cs:行号 67

fangguangyang, 2019-05-10:
[图片]

fangguangyang, 2019-05-10:
[image]

fangguangyang, 2019-05-10:
[image]
I chose to wipe win10m for the first time.  now,I can not deploy for the second time

tkiapril, 2019-05-11:
0. Install WDRT if you haven't already.
1. Open WPInternals.
2. Go to Manual Mode link.
3. Click inturrupt the boot process link on the right panel.
4. Connect your 950XL to the host and hold the power button until it vibrates and then release the power button.
5. Close WPInternals.
6. Open WDRT and press the button that says my device is not detected. Press LUMIA.
7. Follow the process and restore your phone. If you get a red screen with two arrows pointing to each other on it, relaunch WDRT and repeat the process from 6.
8. Once you get your W10M back, go through the activation process and fully shut it down and boot twice.
9. Unlock the bootloader and enable Mass Storage Mode.
10. Before you try to deploy WoA again, fully remove the deployer folder, download the newest release once again (because your copy might be corrupted), extract it and retry.

BaRRaKudaRain, 2019-05-12:
Good advice!

Open WPInternals. 
Go to Manual Mode link. 
Close WPInternals.

SuperJMN, 2019-05-16:
@fangguangyang Have you solved this problem? It seems your phone doesn't have a required file. What phone model are you try to deploy to?

fangguangyang, 2019-05-16:
> @fangguangyang Have you solved this problem? It seems your phone doesn't have a required file. What phone model are you try to deploy to?

Solved, thank you for your help.

closed 2019-05-16 by fangguangyang