WOA-Project/WOA-Deployer-Lumia #32: Suggestions : add some chinese donation channels, eg. ali pay or wechat pay, to this Projection

qindiandadudu, 2019-05-13:
Developers of WOA projection,
Good Job! Thanks for you or everyone of projection. 
I'm a Microsoft fun. Actually, I also have a Lumia 950XL and it has been sleeping in a drawer for years. It's WOA projection brought it to light. So I'm going to donate some heart (money) to it. Unfortunately, I don't have a Paypal account. Instead of it，aili pay and wechat pay is popular in china. If you add those donation channels to this projection, it will receive more donations. In fact, many chinese friends alse want to contribute to this project, both in terms of work and financial resources. 
Finally, thanks again to you works.
Sincerely,

CornyjK, 2019-05-13:
A question: Is any of these services: available in English, free to use and does not need to be Chinese (eg. have Chinese number)?

qindiandadudu, 2019-05-13:
Visite this Url [https://memberprod.alipay.com/account/reg/index.htm](url) and use google translate to convert the language it into English! good luck.

CornyjK, 2019-05-13:
I would do this, but I asked the core team no one seemed to be interested at all, not sure what could you buy with ali pay anyways.

qindiandadudu, 2019-05-14:
Just because doing this can let all the people who want to donate money to this projection don't have to sign up to paypal. If you think doing this no help, that's ok. Take it easy.

SuperJMN, 2019-05-16:
@qindiandadudu I would like to do it, but it seems impossible to get real money from Ali Pay. They act like islands that don't want to share the money with the rest of the world.

qindiandadudu, 2019-05-17:
@SuperJMN This projection is a Good Job! I have repeated windows 10 on arm in my raspberry 3 model B+. The next step is repeat it on my Lumia 950 XL.

I'm sorry for overestimated ali pay. In the future, I will sign up for a PayPal account to support this projection.

closed 2019-05-17 by qindiandadudu