WOA-Project/WOA-Deployer-Lumia #33: "Syntax error: unexpected end of input, expected comment or identifier." when trying to deploy Windows 10 for ARM build 18898

tamto33, 2019-05-16:
**Describe the bug**
After hitting the Deploy button, WoA Deployer reports: "Syntax error: unexpected end of input, expected comment or identifier."

**To Reproduce**
Steps to reproduce the behavior:
Deploy Windows 10 for ARM build 18898 using latest WoA Deployer

**Expected behavior**
Deploying the Windows normally

**Screenshots**
[image]

**Log file**
[Log-20190516.txt](https://github.com/WOA-Project/WOA-Deployer-Lumia/files/3185587/Log-20190516.txt)

SuperJMN, 2019-05-16:
OK, check this folder inside WOA Deployer: 
"Scripts\Talkman" 
Then, open the file SingleSim.txt. Is it empty? what does it contain?

SuperJMN, 2019-05-16:
Fixed with https://github.com/WOA-Project/WOA-Deployer-Lumia/commit/3e85e12276129797d57abdfa960436b4f1faf7b0

closed 2019-05-16 by SuperJMN