WOA-Project/WOA-Deployer-Lumia #37: [Error] Phone disk preparation failed. The partition size cannot be negative: -9.77 GB

cikeZ00, 2019-05-28:
**Describe the bug**
WOA tries to resize the data partition to "-9.77 GB"
which ofc leads to the "Phone disk preparation failed" error

**Expected behavior**
WOA resizes the partition normally and the installation proceeds

**Log file**
[Log-20190528.txt](https://github.com/WOA-Project/WOA-Deployer-Lumia/files/3229890/Log-20190528.txt)

EbrahimBGA, 2019-06-01:
The same issue,
[image]

SuperJMN, 2019-06-15:
Check with DISKPART the partition layout you have. Disk Management tool doesn't show some types os partitions that may be "hidden" and could prevent you from the deploying.

cikeZ00, 2019-06-26:
Update on the whole thing.

So i managed to get it to work by resetting all the partitions back to how it was on stock.
Easiest way to fix this issue is to flash the stock FFU image and then deploy.
So im guessing its because of the old deployments. or possibly the drive letters changing, because i did this on a fresh install of windows.

closed 2019-06-26 by cikeZ00