WOA-Project/WOA-Deployer-Lumia #38: Error: Phone disk preparation failed. Cannot prepare the phone for the deployment

Kethst, 2019-06-08:
**Describe the bug**
Using v2.5.5 of WOA Deployer and I have tried Windows 10 Pro builds 18895 and 18865 multiple times on a Lumia 950XL. Each time it fails with this error. "Phone disk preparation failed. Cannot prepare the phone for the deployment"

It opens three drives before it fails, which are the following: MainOS, Local Disk and a partition called EFIESP. The log says there's only 38 MB available.
[image]

Any suggestions?

**Log file**
[Log-20190607.txt](https://github.com/WOA-Project/WOA-Deployer-Lumia/files/3268061/Log-20190607.txt)

Kethst, 2019-06-08:
I managed to get it to work finally. I had to use the Windows Device Recovery Tool to put down a clean image of the Windows Phone OS. Once I had the clean image down, I went through the process again and it worked. 

One thing to note, the message box that popped up about injecting the drivers said for Lumia devices, to use v2.6 of the WoADeployer tool. The latest version I could find was only v2.5.5

feherneoh, 2019-06-11:
The undersized ESP looks like something caused by unlocking using an old WPInternals build

Just ignore the message about v2.6, it doesn't exist

Flem100, 2019-07-24:
I have the same problem, not enough free space, and I don't know how to remove the old woa version. I have now tried to use the Windows Device Recovery Tool, which sounds like, it's going to do, what I need :).
It took only 5 minutes.

feherneoh, 2019-07-25:
you are better off using WPInternals, as it relocks the bootloader before flashing, and flashing over an unlocked bootloader might cause some problems

Flem100, 2019-07-25:
hey @feherneoh thx for the warning :).
But using WP Internals requires an FFU file and I don't have one for Windows Mobile. There seems to be different places to get one, but all the posts are old. Do you have a preference?
I successfully installed WOA and it's awesome. I just didn't know that the cellular stuff had been "cancelled", which was the main reason to update my WOA and try to use it as a phone OS.
I am thinking to use an old build, before Microsoft killed it. Is that what you guys do?

feherneoh, 2019-07-25:
WPInternals can download the ffu file for you if you use a recent version, no need to look for it yourself

The latest retail build still has cellular working, it was only removed in recent insider builds
I would suggest you to use the latest retail 1903 build, being 18362.239 at the moment

Flem100, 2019-07-25:
Thank you very much @feherneoh !
I ran 18941.1001.190713 which was the latest one, yesterday, so I should try the one you mentioned.
I know I wouldn't be able to sms, but what about the ARM Phone App? Does it work, it didn't work in the version I tried, but that makes sense, since the cellular wasn't available.
Well it showed some error next to the cellular logo in the network section, I forgot what it said.

feherneoh, 2019-07-26:
on 1903 Phone app is deployed together with the system, and works too
You can also receive SMS messages, but the built-in app does not allow you to send them

Flem100, 2019-07-26:
@feherneoh That's awesome thanks!
Last question: I have installed WOA twice. The first time was back in February when the charging didn't work, but at least I could have Windows mobile installed as well so I could switch between the OS. This time, it seemed it was no longer possible. Is that true, or am I doing something wrong? As far as I understand, I can still choose to keep Windows 10 Mobile, in the WOA Deployer 2.6, but it didn't work.
I was thinking to make a youtube clip to explain and show how to install WOA and what to expect, it seems the current guides are somewhat outdated. I am wrong that the project has gone a little out of steam, since Microsoft removed the celluar stuff. From what I have read, Gus wants to work on something else now.
Should I be worried about any legal stuff?
Thanks in advance! :)

feherneoh, 2019-07-26:
To keep W10M too:

Deploy WoA as usualy, reserving some space for it
Boot it up, set some stuff you need up, INSTALL ANY UPDATE YOU NEED
Reboot into mass storage, open the deployer, and on the dualboot tab enable dualboot

Note that you'll have to disable dualboot again to be able to install updates on WoA

AlairEileen, 2019-09-04:
[image]
[image]

Log Info:
2019-09-04 18:38:17.002 +08:00 [Verbose] Started WOA Deployer for Lumia v2.6
2019-09-04 18:38:20.548 +08:00 [Verbose] Checking path: "H:\sources" exists
2019-09-04 18:38:23.863 +08:00 [Verbose] Default directory for WimFolder has been set to "H:\sources"
2019-09-04 18:38:23.874 +08:00 [Verbose] Trying to load WIM metadata file at '"H:\sources\install.wim"'
2019-09-04 18:38:23.895 +08:00 [Verbose] Getting WIM stream
2019-09-04 18:38:23.896 +08:00 [Verbose] (WIM) Finding Magic Bytes...
2019-09-04 18:38:23.896 +08:00 [Verbose] (WIM) Found Magic Bytes at 0
2019-09-04 18:38:23.897 +08:00 [Verbose] (WIM) Finding WIM XML Data...
2019-09-04 18:38:23.897 +08:00 [Verbose] (WIM) Found WIM XML Data at 042979272732
2019-09-04 18:38:23.919 +08:00 [Verbose] Wim metadata deserialized correctly WimMetadata { TotalBytes: "4297927273", Images: [ImageMetadata { DirectoryCount: "23607", FileCount: "101161", TotalBytes: "16814994097", HardLinkBytes: "6419216843", CreationTime: Time { HighPart: "0x01D562E1", LowPart: "0x11507570" }, LastModificationTime: Time { HighPart: "0x01D562E1", LowPart: "0xBF864018" }, WimBoot: "0", Windows: Windows { Arch: "12", ProductName: "Microsoft® Windows® Operating System", EditionId: "Professional", InstallationType: "Client", ServicingData: ServicingData { GdrDuRevision: "0", PKeyConfigVersion: "10.0.18362.1;2016-01-01T00:00:00Z" }, ProductType: "WinNT", ProductSuite: "Terminal Server", Languages: Languages { Language: "en-US", Default: "en-US" }, Version: Version { Major: "10", Minor: "0", Build: "18362", SpBuild: "329", SpLevel: "0" }, SystemRoot: "WINDOWS" }, Name: "Windows 10 Pro", Description: "Windows 10 Pro", Flags: null, DiplayName: null, DisplayDescription: null, Index: "1" }] }
2019-09-04 18:38:23.921 +08:00 [Verbose] WIM metadata file at '"H:\sources\install.wim"' retrieved correctly
2019-09-04 18:38:25.045 +08:00 [Information] # Starting deployment...
2019-09-04 18:38:28.143 +08:00 [Debug] Device Stream disposed
2019-09-04 18:38:28.210 +08:00 [Information] 
2019-09-04 18:38:28.210 +08:00 [Information] # Fetching Deployment Scripts
2019-09-04 18:38:28.266 +08:00 [Information] Fetching from GitHub: https://github.com/WOA-Project/Deployment-Scripts - master
2019-09-04 18:38:30.519 +08:00 [Debug] Saving metadata { Repository: RepoInfo { Owner: "WOA-Project", Repository: "Deployment-Scripts" }, Commit: "a36c0d1a0b3451312a0223fa088b1e5f5e49f4e3", Branch: "master", DownloadedOn: 09/04/2019 18:38:29 +08:00 }
2019-09-04 18:38:31.490 +08:00 [Debug] Initializing Device Stream...
2019-09-04 18:38:31.503 +08:00 [Debug] Reading device GPT...
2019-09-04 18:38:31.505 +08:00 [Debug] Found DPP
2019-09-04 18:38:31.505 +08:00 [Debug] Found SBL1
2019-09-04 18:38:31.505 +08:00 [Debug] Found PLAT
2019-09-04 18:38:31.506 +08:00 [Debug] Reading SBL1 Partition
2019-09-04 18:38:31.506 +08:00 [Debug] Starting reading operation
2019-09-04 18:38:31.506 +08:00 [Debug] Reading from 54525952 to 55574016
2019-09-04 18:38:32.104 +08:00 [Debug] Reading pconf.bin from file system...
2019-09-04 18:38:32.140 +08:00 [Debug] Reading product.dat from file system...
2019-09-04 18:38:32.147 +08:00 [Error] An error has occurred
System.DivideByZeroException: Attempted to divide by zero.
   at DiscUtils.Fat.FatFileSystem.DetectFATType(Byte[] bpb)
   at DiscUtils.Fat.FatFileSystem.Initialize(Stream data)
   at DiscUtils.Fat.FatFileSystem..ctor(Stream data)
   at Deployer.Lumia.NetFx.PhoneMetadata.PhoneInfoReader.GetPhoneInfo(UInt32 diskNumber) in D:\a\1\s\Source\Deployer.Lumia.NetFx\PhoneMetadata\PhoneInfoReader.cs:line 130
   at Deployer.Lumia.NetFx.PhoneMetadata.PhoneModelInfoInfoReader.GetPhoneModel(UInt32 diskNumber) in D:\a\1\s\Source\Deployer.Lumia.NetFx\PhoneMetadata\PhoneModelInfoInfoReader.cs:line 17
   at Deployer.Lumia.Phone.<GetModel>d__7.MoveNext() in D:\a\1\s\Source\Deployer.Lumia\Phone.cs:line 37
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Deployer.Lumia.WoaDeployer.<RunDeploymentScript>d__13.MoveNext() in D:\a\1\s\Source\Deployer.Lumia\WoaDeployer.cs:line 67
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Deployer.Lumia.WoaDeployer.<Deploy>d__11.MoveNext() in D:\a\1\s\Source\Deployer.Lumia\WoaDeployer.cs:line 41
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Deployer.Lumia.Gui.ViewModels.DeploymentViewModel.<Deploy>d__11.MoveNext() in D:\a\1\s\Source\Deployer.Lumia.Gui\ViewModels\DeploymentViewModel.cs:line 68
2019-09-04 18:38:32.164 +08:00 [Information] Error: Attempted to divide by zero.
2019-09-04 18:39:00.830 +08:00 [Information] 
2019-09-04 18:50:56.861 +08:00 [Verbose] Started WOA Deployer for Lumia v2.6
2019-09-04 18:51:41.598 +08:00 [Verbose] Getting Dual Boot Status...
2019-09-04 18:51:44.275 +08:00 [Debug] Device Stream disposed
2019-09-04 18:51:45.106 +08:00 [Debug] Device Stream disposed
[... cut ...]
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Deployer.Lumia.Gui.ViewModels.DeploymentViewModel.<Deploy>d__11.MoveNext() in D:\a\1\s\Source\Deployer.Lumia.Gui\ViewModels\DeploymentViewModel.cs:line 68
2019-09-04 21:05:27.154 +08:00 [Information] Error: Could not locate file
2019-09-04 21:05:39.554 +08:00 [Information]