WOA-Project/WOA-Deployer-Lumia #48: CPU speed reduced on both 950 and 950 XL

falkor2k15, 2019-10-02:
Under Windows 10 System, 950 is only showing as 0.6 GHz. My 950 XL started out as 1.94 GHz, but since installing WOA a few times it's now reduced to 1.55 GHz.

Please could you advise? It's claiming that the 950 has 6 processors and the 950 XL has 8 processors; not sure if that part is right either.

Note: this happens on all versions of Windows 10 - 1709 through to 1903.

After reimaging from one version to another is there a special process I need to follow to properly clean up the old deployment - and one that might also fix this CPU clock speed issue?

falkor2k15, 2019-10-02:
FALSE ALARM! Somebody else already confirmed that the big cores are not working in the 950. As for the XL, I just checked more detail under CPU and it still lists the base speed as 1.96 GHz and fluctuates above that of the small cores.

SuperJMN, 2019-10-02:
Anyways, this issue doesn't have anything to do with WOA Deployer ;) Glad you figured out. The speed is fluctuating due to power management. I'm afraid I can't help you there.

closed 2019-10-02 by SuperJMN