WOA-Project/WOA-Deployer-Lumia #53: Red de telefonía móvil: "La cuenta no está activa"

CARLOSTORTAJADA, 2019-10-23:
He utilizado Woa Deployer versión 2.6 para instalar Windows 10 pro ARM en un Lumia 950 XL con todas las instrucciones y comprobando que se han cargado todos los drivers de Cityman, sin embargo sigue sin funcionar el CELULAR, a pesar de que en STATUS se indica que funciona tanto para datos como para llamadas.
Cuando se abre el celular indica "teléfono desactivado" y en redes Red de telefonía móvil: "La cuenta no está activa"
Lo he instalado en varias ocasiones y se ha reiniciado en cada una de ellas también múltiples veces sin obtener éxito hasta el momento, por supuesto que tiene una SIM activa.

tymcarlos, 2019-10-30:
Puesto que la tarjeta SIM si que la detecta, es lógico pensar que es debido a la compañía de telefonía, la que aparece en las imágenes de imgur.com son de SFR (Francia) y mi compañía es movistar (España).

tymcarlos, 2019-11-06:
Realizados los mismos pasos con el Lumia 950 obtengo el mismo resultado, tanto la red de telefonía móvil como el teléfono (Celular) dicen cuenta no activa o desactivado. Sigue sin haber conexión telefónica cuando se inicia en Windows 10 Arm.

CornyjK, 2019-11-07:
English ONLY.

closed 2019-11-07 by CornyjK