WOA-Project/WOA-Deployer-Lumia #55: Mobile phone network: "The account is not active"  [question] [tech support required]

tymcarlos, 2019-11-07:
I have used Woa Deployer version 2.6 to install Windows 10 pro ARM on a Lumia 950 XL with all the instructions and checking that all Cityman drivers have been loaded, however the CELLULAR still does not work, although in STATUS it is indicated that It works for both data and calls.
When the cell phone is opened it indicates "phone deactivated" and in networks Mobile phone network: "The account is not active"
I have installed it several times and it has been restarted in each of them also multiple times without obtaining success so far, of course it has an active SIM.
Performed the same steps with the Lumia 950 I get the same result, both the mobile phone network and the phone (Cellular) say account is not active or deactivated. There is still no phone connection when it starts in Windows 10 Arm.
Since the SIM card does detect it, it is logical to think that it is due to the telephone company, the one that appears in the images of imgur.com are from SFR (France) and my company is movistar (Spain).

gus33000, 2020-03-28:
Not related to woa deployer, closing. (Issue also got fixed on the driver front of things ever since)

closed 2020-03-28 by gus33000

tymcarlos, 2020-03-28:
Rotten lie

SuperJMN, 2020-03-28:
@tymcarlos, it's not a lie. It has nothing to do with WOA Deployer. Sorry.

CornyjK, 2020-03-28:
😂😂😂