WOA-Project/WOA-Deployer-Lumia #58: How to enable dual boot in 2 clicks (lumia 950 xl)

EdOverride, 2020-03-14:
I had Lumia 950 XL with windows 10 mobile.

I successfully installed windows 10 arm on Lumia 950 xl, using this tool, which is fine, but now I'm looking at how to dual-boot it with Windows 10 mobile.

The page said "Using this tool, you can enable Dual Boot in literally 2 clicks, but where are those clicks?
[image]

Don't see any option like that.

I also found this issue [#16](https://github.com/WOA-Project/WOA-Deployer-Lumia/issues/16), which basically asks the same thing.

I tried it mentioned methods, but the first one, which is:
* Switch to Mass Storage Mode
* Go to [MainOS]\EFIESP\EFI\Microsoft\BOOT\BCD
* Execute:
   ```
   bcdedit /store BCD /display order {7619dcc9-fafe-11d9-b411-000476eba25f} /addlast
  ```

Gave me error:
```
F:\EFIESP\EFI\Microsoft\BOOT>bcdedit /store BCD /display order {7619dcc9-fafe-11d9-b411-000476eba25f} /addlast
An unknown command was specified.
Run "bcdedit /?" for command line assistance.
```

And second method also, which is:
* Reboot to mass storage mode
* Open DISKPART
* Like the previous time you have to "select" your disk and partition with diskpart
* LIST DISK
* SELECT DISK x (where x should be the number of the disk which represents your phone)
* LIST PARTITION
* SELECT PARTITION y (where y should be the number of the “WOABOOT” partition we created for the Windows installation, which we assigned the "S:" drive letter)
* Then set the type of the partition that identifies it as a "Microsoft basic data partition" with the following command (still inside DISKPART)
* Set id=ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
* Reboot
* Dual boot is now working

Also doesn't work, as there is no partition with name `WOABOOT`:
[image]

That issue was resolved by:
`The trick is to disable dual boot (via WOA Deployer) and then again enable it again`.

But again, I don't see any option like that in the tool.

So, how does one enable `Windows 10 arm` and `Windows 10 mobile` dual-boot on Lumia 950 XL?

EdOverride, 2020-03-14:
I did some checking on how the `bcdedit` command works ([link](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/adding-boot-entries)) and turns out that issue had typo.

The command should have been:
```
bcdedit /store BCD /displayorder {7619dcc9-fafe-11d9-b411-000476eba25f} /addlast
```

However, this just moves around existing boot items, namely "Dummy, please ignore". But selecting it, device just reboots after 5sec of black screen.

So again, how does one enable dual-boot between `windows 10 arm` and `windows 10 mobile` ?

gus33000, 2020-03-28:
Click on the dual boot text in the application itself, you'll get to the corresponding page with a button to check dual boot availability

gus33000, 2020-03-28:
[image]
here

AMBNETO, 2020-04-12:
In advanced menu I checked the box "Keep Windows 10 Mobile", or something like that, and then started deploying. After finishing the process, I clicked on "Force Dual Boot" in advanced menu. That's pretty much it, I got dualboot working. If you click on "check dual boot status", you will still get an error message.

LeoMaath, 2020-11-05:
Horay. After doing like AMBNETO indicated the deployer does not seem to act. The page is frozen. And switch back to the page Deployment or Dual Boot is not possible.
But after some 5 minutes the program displayed the dual boot has been successfully arranged. Perhaps a little warning can be added, like 'take a coffee, or take the time to buy me a coffee'

eoegithub1, 2020-11-14:
Advanced menu is greyed out! Could only click on Deployment.

timmynana, 2021-01-27:
Thanks to you all. I had the exact same issue and confirm that making sure the phone was in Mass Storage mode and using WOA Deployer to check if dual boot mode was enabled, I was able to dual boot. It took around 20 seconds for me so I must have been fortunate. No command prompt work necessary. Thanks!