WOA-Project/WOA-Deployer-Lumia #59: Please make this project More better using window 10x images

fisforfaheem, 2020-03-27:
Please use the "Preview" tab above to view this message if you are seeing this in the new issue text box.

Please note that this is a **WoA Deployer** issue tracker, use this issue tracker only for the WoA Deployer related enhancements.

Please open ticket in [WOAProject forum](https://www.woaproject.net/viewforum.php?f=17) for anything else, such as:

- Reporting bugs of other tools, drivers, etc.
- Asking for help
- Asking about project progress.

# Please remove the text above before opening an issue.

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

gus33000, 2020-03-28:
Windows 10X is an i686 amd64 x86_64 Operating System meaning it is therefore incompatible with an armv7a/armhf/armv8a/armv8 CPU/SOC

closed 2020-03-28 by gus33000

fisforfaheem, 2020-04-02:
please dosomthign for lumia822

n

wrote: