WOA-Project/WOA-Deployer-Lumia #61: How to upgrade

biconix, 2020-04-12:
A while ago i installed wind10 on my 950XL. i did not set up a dual boot.
I assume that there has been an upgrade in the past year+.
I would like to upgrade the device to a more current version as this one has not been working well. In fact I cannot charge it properly. and it can't get past the login screen
What steps should I do in order to do the upgrade?
if i cannot upgrade is there a way to restore the original Windows phone OS?
(or is the phone bricked?)

CornyjK, 2020-04-12:
You can only redeploy WOA or reflash stock via WDRT

closed 2020-04-12 by CornyjK

biconix, 2020-04-13:
I have tried that. 
first, WDRT doesn't detect the device. (Device could not be found) even if I select the button for Lumia Windows Phone
Second because the phone cannot be charged by USB, I have to charge it wirelessly at the same time it is plugged in to my Surface via USB
Third I cannot successfully log in anymore. I am able to enter my PIN but i can't get past the spinning welcome. 
So, is this bricked? or is there a way to update the version of windows on the phone?