WOA-Project/WOA-Deployer-Lumia #62: refind not working

blackdiamont, 2020-04-14:
Hi there,

I've just read that refind is now included and I'd like to give it a try for linux booting.
But if I start the phone and boot Windows10 where the message appears to press power oder tap ... nothing happens and it's just booting Windows ... did I miss something?

AppleInternal, 2024-06-27:
> Hi there,
> 
> I've just read that refind is now included and I'd like to give it a try for linux booting. But if I start the phone and boot Windows10 where the message appears to press power oder tap ... nothing happens and it's just booting Windows ... did I miss something?

What your Lumia?

blackdiamont, 2024-07-24:
It's a Lumia 950 XL but I havn't used it for a while.

AppleInternal, 2024-07-26:
> It's a Lumia 950 XL but I havn't used it for a while.

its time to use it, refind will work if you flash WoA FFU or Windows 11X FFU by me