WOA-Project/WOA-Deployer-Lumia #69: Add lumia 640 support

fisforfaheem, 2020-08-26:
Millions still use it and it will run the windows 10 please do it

DremOSDeveloperTeam, 2020-08-31:
Not only is this probably impossible, but it's even more underpowered than the Lumia 950 so it would be unbearable if it ever worked at all.

fisforfaheem, 2020-09-02:
Atleast make an app to revive these older devices by throwing in some performanc eimproving hacks please

DremOSDeveloperTeam, 2020-09-03:
One does not simply "make an app" that can "throw in some performance improving hacks". The closest thing to what you want is a [custom ROM](https://forum.xda-developers.com/windows-10-mobile/th2-restone-windows-10-custom-roms-t3440144/amp/) but I have no clue if there is a custom ROM for your phone or if they would help performance. This, however, is out of the scope of this project.

fisforfaheem, 2020-09-04:
Oh k! bro Please Please Do something great! and show us work hard and release!

closed 2020-11-11 by cewbdex