WOA-Project/WOA-Deployer-Lumia #73: Is it possible to update Windows as on PC? (via Microsoft Update)

chiaravalle, 2020-12-01:
Hello everyone.
As in subject, I wonder if is it possible to update WoA using Microsoft Update, just in case a newer version comes up.
Is it possible, or am I required to download and flash every time?

Thanks in advance for any help.

SuperJMN, 2020-12-01:
It's possible as long as you disable Dual Boot before the updating (the dual boot option that enables Windows 10 Phone + WOA).

closed 2020-12-01 by SuperJMN

chiaravalle, 2020-12-01:
Thanks a lot

chiaravalle, 2020-12-02:
Dumb question: when I installed, I didn't made it dual boot.
Am I required to tweak something anyhow?

Thanks in advance.

SuperJMN, 2020-12-02:
No, if you don't touch anything, it should update just right.

Obtener Outlook para Android<https://aka.ms/ghei36>

chiaravalle, 2020-12-02:
Ok, thanks a lot more

chiaravalle, 2020-12-11:
One last question: does updating break driver compatibility? I'm running build 18363, if I update, will drivers quit working?

chiaravalle, 2020-12-11:
Nevermind, I just registered in the forums, and I'll look there for answers.
Please, discard my messages.