WOA-Project/WOA-Deployer-Lumia #74: Driver update - reflash?

chiaravalle, 2020-12-02:
Hello everyone.
Is it possible to update WoA drivers without flashing the image again?
If so, how?

Thanks in advance for any help.

chiaravalle, 2020-12-02:
P.S.: I need to manually install GPU and Mobile Phone Drivers, where can I find them?
And how can I install them?
I'm using Latest Public Release Build

SuperJMN, 2020-12-02:
I'm sorry, we're out of luck. Because the way that drivers are installed and their interdependencies, it's not possible to reinstall/update the drivers. You need to flash/redeploy.

closed 2020-12-02 by SuperJMN

chiaravalle, 2020-12-03:
Ok, which image should I flash to get them?

chiaravalle, 2020-12-03:
I guess the drivers are the same, even on most recent builds. Aren't they? If so, I won't need to reflash, correct?

chiaravalle, 2020-12-04:
Please, tell me

chiaravalle, 2020-12-11:
Nevermind, I just registered in the forums, and I'll look there for answers.
Please, discard my messages.