WOA-Project/WOA-Deployer-Lumia #79: When I click the "Deploy" botton and waited a while,It said "Try dividing by zero" to me

i35010u, 2021-07-08:
[image]
the Chinese words in the image means "Try dividing by zero"
I deleted the "Downloaded" folder,delete the Deployer and redownload the newest Deployer from GitHub,but I still didn't solve this problem.

i35010u, 2021-07-08:
[image]

i35010u, 2021-07-08:
I also changed almost each of these settings

i35010u, 2021-07-08:
[Log-20210708.txt](https://github.com/WOA-Project/WOA-Deployer-Lumia/files/6782766/Log-20210708.txt)

SuperJMN, 2021-07-08:
That looks to me like your DPP partition is corrupted ☹

[image]<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F60495797%2F124883780-43633c00-e004-11eb-80f0-bbdf61e83c99.png&data=04%7C01%7C%7Cd9e5e928e42d436bc0dc08d941e615ee%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637613278723277249%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=7D9McKYhq2a470e2P6sptQ0HmX8VeKVrGC6zmt7QM%2Bw%3D&reserved=0>

i35010u, 2021-07-11:
I check my phone in Mass Storage Mode,the DDP partition is actually lost....but how can I solve this problem?Can I create a DPP patition and copy the files from another person's phone and just edit the files?

maijiajie, 2021-07-11:
> I check my phone in Mass Storage Mode,the DDP partition is actually lost....but how can I solve this problem?Can I create a DPP patition and copy the files from another person's phone and just edit the files?

我有dpp分区的备份文件！

i35010u, 2021-07-12:
> > I check my phone in Mass Storage Mode,the DDP partition is actually lost....but how can I solve this problem?Can I create a DPP patition and copy the files from another person's phone and just edit the files?
> 
> 我有dpp分区的备份文件！

你可以把备份文件发给我吗？（我之前也试过，但是我复制的是一台工程机上面的DPP分区，貌似没有用
（似乎DPP里面的文件都是二进制文件，并不是文本之类的易于编辑的文件）

maijiajie, 2021-07-12:
> 你可以把备份文件发给我吗？（我之前也试过，但是我复制的是一台工程机上面的DPP分区，貌似没有用
> （似乎DPP里面的文件都是二进制文件，并不是文本之类的易于编辑的文件）
我用github还不是很熟练，所以上传到百度云了
[https://pan.baidu.com/s/1jum544CKrRagsqegYFhS6Q](url)提取码0000
用DG或者PG之类的硬盘管理软件还原dpp分区
然后我就可以刷进去了
虽然我的刷进去还是提示引导出了问题
0xc0000428
不清楚你的刷进去能不能开

i35010u, 2021-07-12:
> I check my phone in Mass Storage Mode,the DDP partition is actually lost....but how can I solve this problem?Can I create a DPP patition and copy the files from another person's phone and just edit [...]

谢谢！我去试试了

i35010u, 2021-07-12:
> I check my phone in Mass Storage Mode,the DDP partition is actually lost....but how can I solve this problem?Can I create a DPP patition and copy the files from another person's phone and just edit [...]

我发现这东西的机型是存在DPP里面的。。。所以不能跨机型/RM刷DPP。。。。（刷完了我手机原来的RM-1116变成了RM-1104。。。可能我还是得找一台RM-1116的手机来复制

SuperJMN, 2021-07-13:
Hi! Could you please switch the conversation to English so we can take part into it, just in case? Thanks :)

i35010u, 2021-07-17:
> Hi! Could you please switch the conversation to English so we can take part into it, just in case? Thanks :)

OK,I will try again on a English system soon.I'm going to switch the language of Windows.

maijiajie, 2021-07-20:
> 我发现这东西的机型是存在DPP里面的。。。所以不能跨机型/RM刷DPP。。。。（刷完了我手机原来的RM-1116变成了RM-1104。。。可能我还是得找一台RM-1116的手机来复制

那个文件名我标明了是950 rm1104的 ，我的1105刷这个也是不行