WOA-Project/WOA-Deployer-Lumia #80: Excepção de HRESULT: 0x80070005 (E_ACCESSDENIED))

phgeraldes, 2021-07-12:
Hi there im trying to install windows 10 on lumia 950xl but without luck, here´se the steps that i made:

1 - download windows 10 on ARM ISO - done
2 - Unlocking Lumia 950 bootloader with wpinternals - done
3 - Installing Windows 10 on ARM on Lumia 950 - problem

Let me described i downloaded woa deployer for lumia 2.7.1 search for the arm iso that i mounted yearlier the install wim, first time everything was running fine until it appears that it should 3x faster to mount on the cellphone the mass storage, please click the button until it show developer tool and then mount mass storage, I done that and after nothing works, the windows mobile does not boot and i only have access to developer tools to put it on uefi mass storage.

When i try to deploy again error shows

Excepção de HRESULT: 0x80070005 (E_ACCESSDENIED))

I only have two partition one of 3gb an other of 32mb

Can someone help here to fix this

In attach i send the log file
[Log-20210712.txt](https://github.com/WOA-Project/WOA-Deployer-Lumia/files/6802050/Log-20210712.txt)

Thanks

phgeraldes, 2021-07-12:
Update, try once more and is giving error em Deployer.Lumia.Gui.ViewModels.DeploymentViewModel.<Deploy>d__11.MoveNext() em D:\Repos\WOA-Project\WOA-Deployer-Lumia\Source\Deployer.Lumia.Gui\ViewModels\DeploymentViewModel.cs:line 67
2021-07-12 15:28:06.051 +01:00 [Information] Error: Phone disk preparation failed. Cannot prepare the phone for the deployment
2021-07-12 15:28:16.296 +01:00 [Information]

CHEIKH33, 2021-07-12:
Very good

phgeraldes, 2021-07-13:
Does anyone can help here, i already restore the lumia 950xl with windows mobile, try again and same problem, right now i have 3 partitions, but problem remains.