WOA-Project/WOA-Deployer-Lumia #85: Is this normal?

TyTytheFurKit, 2021-09-11:
I've downgraded to Windows 7 recently, and I decided to see if there would be any different results when connecting the Lumia 950 i have to my computer while Windows 10 on arm was booted, there was a MAJOR difference, I can access the SD Card I have in there directly and the MainOS Partition from the Windows 10 Mobile OS, and it will live update both when I make changes from the phone or the computer I use, and it uses a Windows 11 on Lumia icon for when it's booted into that mode.
[image]
[image]
Is this expected behavior? I'm just really amazed by this and I want to be sure that by it doing this it isn't damaging the phone in anyway.

TyTytheFurKit, 2021-09-11:
[image]
It even has the version number inside the Device Properties Dialog

SuperJMN, 2021-09-11:
Yes, that's normal. You should thank @gus33000 for that :)

closed 2021-09-11 by SuperJMN