WOA-Project/WOA-Deployer-Lumia #90: JUNK JUST PURE JUNK I'M SO MAD AN ENTIRE DAY WASTED I WAITED FOR SO MANY HOURS USING UUPDUMP TWICE AND THEN JUST USING AN OFFICIAL .iso FROM MICROSOFT'S MEDIA CREATION TOOL AND THE EXACT SAME THING I SELECT THE install.wim FROM UUPDUMP OR THE install.esd FROM THE OFFICIAL .iso AND IT IMMEDIATELY FLAT OUT REFUSES EVERY TIME WITHOUT FAIL AND THROWS A BS ERROR "Invalid WIM file"  "Could not read the metadata from the WIM file. Please, check it's a valid .WIM file" AND THE ONLY OPTION IS "OK" IT DOESN'T EVEN TRY

44cK3r, 2023-09-19:
Please use the "Preview" tab above to view this message if you are seeing this in the new issue text box.

Please note that this is a **WoA Deployer** issue tracker, use this issue tracker only for the WoA Deployer related bugs.

Please open ticket in other repositories for anything else, such as:

- Reporting bugs of other tools, drivers, etc.
- Asking for help
- Asking about project progress.

# Please remove the text above before opening an issue.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Log file**
Please put your log file here (just drag&drop it).

closed 2023-09-19 by cewbdex

Xuyiyang23333, 2023-11-04:
Calm down and try this.
https://www.reddit.com/r/windowsphone/comments/15s1mvc/woa_deployer_image_compatibility/
```
DISM /Export-Image /SourceImageFile:[downloaded install.wim] /SourceIndex:1 /DestinationImageFile:[new install.wim destination] /CheckIntegrity
```

44cK3r, 2023-11-04:
> Calm down and try this.
> https://www.reddit.com/r/windowsphone/comments/15s1mvc/woa_deployer_image_compatibility/
> ```
> DISM /Export-Image /SourceImageFile:[downloaded install.wim] /SourceIndex:1 /DestinationImageFile:[new install.wim destination] /CheckIntegrity
> ```

Ok thank you