WOA-Project/WOA-Deployer-Lumia #92: Incompatible with recent images of uupdump

Arturbc, 2024-03-18:
**Describe the bug**
On try deploy a ARM64 ISO on a Lumia 950, on select of install.wim file, appeared a "metadata error" message. I have try download many ISOs, included very old Windows 10 versions I already used before, but never accept.

**To Reproduce**
Steps to reproduce the behavior:

1. Download any uupdump ARM64 ISO;
2. Try deploy;
3. Get a error about metadata invalid.

**Expected behavior**
Is expected accept file install.wim and show a list of edtitions to select and start deploy.

**Screenshots**
[image]
[image]

**Log file**
[Log-20240317.txt](https://github.com/WOA-Project/WOA-Deployer-Lumia/files/14629353/Log-20240317.txt)

zenbaku, 2024-04-04:
+1 also affected by this. Using a WIM validator and mounting the files works, so not sure why woa deployer won't recognize them.

roadrunnerli, 2024-08-18:
+1 also affected. Tried over 10 images from various sources like UUPDUMP or massgrave.dev/windows_arm_links, did not find a single one that works.
Looking mainly for aworking 18363 iso (to preserve the ICan0 info, required for phone call functionality).

NikeDudeeGAMER, 2024-09-01:
> +1 also affected. Tried over 10 images from various sources like UUPDUMP or massgrave.dev/windows_arm_links. I did not find a single one that works. Looking mainly for a working 18363 .iso (to preserve the ICan0 info, required for phone call functionality).
> 
> Found a workaround -> Export the .wim-file from the .wim-file :D DISM /Export-Image /SourceImageFile:[path to invalid install.wim] /SourceIndex:1 /DestinationImageFile:[path to new install.wim] /CheckIntegrity Will correct the issue

where do i type that in

roadrunnerli, 2024-09-05:
> ****where do i type that in****

Type it In a (Windows 11 elevated) command prompt!
From the begin:
First, double-click the 'Iso' that you have created with UUPDump -> that 'mounts' the ISO image as a new drive in explorer. Lets assume that you have a temp folder on your C-Drive (C:\Temp) and the  new drive gets the letter "F:" assigned

Then open a 'CMD' (run as administrator) and type in the command (on one line / no LF)

DISM /Export-Image /SourceImageFile:"F:\sources\install.wim" /SourceIndex:1 /DestinationImageFile:"c:\Temp\install.wim" /CheckIntegrity

Yyou will find the working install.wim in C:\Temp

thanone2, 2025-02-21:
> DISM /Export-Image /SourceImageFile:[path to invalid install.wim] /SourceIndex:1 /DestinationImageFile:[path to new install.wim] /CheckIntegrity

you are a life saver, tysm

burd1, 2025-04-05:
Same error till now, @roadrunnerli has the solution that works, thank you.